chunhuajiang/esp-idf-zh #19: [AliOS-Things入门] 在Linux下搭建开发环境 | IoT-Fans

tidyjiang8, 2017-10-25:
http://iot-fans.xyz/2017/10/25/alios/env/

Cheng-SG, 2017-10-27:
这一行命令不对吧：
echo "/opt/gcc-arm-none-eabi-5_4-2016q3/bin/" >> ~/.bashrc && source ~/.bashrc

你是想写成：
echo "export PATH=$PATH:/opt/gcc-arm-none-eabi-5_4-2016q3/bin/" >> ~/.bashrc && source ~/.bashrc
或者
echo "export TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-5_4-2016q3/bin/" >> ~/.bashrc && source ~/.bashrc
吧？

tidyjiang8, 2017-10-27:
@Cheng-SG
多谢提醒

notagoodman, 2017-12-24:
按照上述步骤编译，出了问题。
lxf@lxf-K43SA:/opt/AliOS-Things$ sudo aos make helloworld@linuxhost
Build AOS Now
Compiling auto_component
Making out/helloworld@linuxhost/libraries/auto_component.a
Making helloworld@linuxhost.elf

Traceback (most recent call last):
  File ".//build/scripts/map_parse_gcc.py", line 15, in <module>
    mem_config_text = re.findall('Memory Configuration\n\nName             Origin             Length             Attributes\n([\s\S]+)\nLinker script and memory map', s)[0]
IndexError: list index out of range
build/aos_target_build.mk:265: recipe for target 'out/helloworld@linuxhost/binary/helloworld@linuxhost.elf' failed
make[1]: *** [out/helloworld@linuxhost/binary/helloworld@linuxhost.elf] Error 1
build/Makefile:197: recipe for target 'main_app' failed
make: *** [main_app] Error 2
[aos] ERROR: "/" returned error code 2.
[aos] ERROR: Command "/ o p t / A l i O S - T h i n g s / b u i l d / c m d / l i n u x 6 4 / m a k e   H O S T _ O S = L i n u x 6 4   T O O L S _ R O O T = / o p t / A l i O S - T h i n g s / b u i l d   - f   b u i l d / M a k e f i l e   h e l l o w o r l d @ l i n u x h o s t" in "/opt/AliOS-Things"